mcore1976/cc1101-tool #1: Would this work for ESP32 Out Of Box ?

jayateertha043, 2023-06-10:
Kindly let me know how I can port this for ESP32 if it doesn't work out of box.

mcore1976, 2023-06-10:
please change following pin numbers to corresponding on ESP32 boardbyte sck = 18;byte miso = 19;byte mosi = 23;byte ss = 5;int gdo0 = 32;int gdo2 = 33;check pin descriptions on the attached imageWysłano z mojego urządzenia Galaxy
-------- Oryginalna wiadomość --------Od: Jayateertha Guruprasad ***@***.***> Data: 10.06.2023  11:20  (GMT+01:00) Do: mcore1976/cc1101-tool ***@***.***> DW: Subscribed ***@***.***> Temat: [mcore1976/cc1101-tool] Would this work for ESP32 Out Of Box ? (Issue #1) 
Kindly let me know how I can port this for ESP32 if it doesn't work out of box.

mcore1976, 2023-06-22:
see the other thread for ESP8266 board

closed 2023-06-22 by mcore1976