mcore1976/cc1101-tool #2: serial terminal issues

woahotter, 2023-06-19:
to start off, this is a super cool project
unfortunately, i can't actually do anything with it because my serial terminal wont actually enter commands

if i type in `help` in putty, the characters dont come up as i type them (sometimes normal for putty), but even if i press enter nothing comes up at all  

i've tried different baud rates, and i'm using a xiao esp32c3 which successfully flashed onto the board with no issues
i've edited gpio too, so even if that is an issue somehow thats out of the question

i feel like this is just user error by me, but either way its worth making an issue for other people

mcore1976, 2023-06-19:
I am using normal Serial Port commands in  ESP32 
 Serial.print(F("Hello world  "));
Assuming you have USB-TTL converter that shoud be hard-wired to one of ESP32 serial ports it should work. However ESP32 has 3 serial ports and these should work only for port that is used for Serial debugging

I am using the same principles as in this example here :
https://deepbluembedded.com/esp32-hello-world-serial-print-arduino/?utm_content=cmp-true

check if this example works for you. 

I am currently waiting for ESP32 board delivery, once I get it I will be able to correct the code.

woahotter, 2023-06-19:
through usb serial i do get the "CC1101 terminal tool connected", but if i do it with a USB-TTL converter i don't get nothing of that, just boot logs and im unable to enter anything

ive used 9600 baud, but i'm currently using 115200 and sticking with that

in the example https://deepbluembedded.com/esp32-hello-world-serial-print-arduino/?utm_content=cmp-true ,  everything worked and the esp32 was able to receive data over usb

**BUT i found out that wasn't the issue**

it seems that it's not receiving anything because its hanging in the `cc1101initialize` function, on [ELECHOUSE_cc1101.Init()](https://github.com/mcore1976/cc1101-tool/blob/93df77a2a169fbe35f637a20343e6607428110cb/cc1101-tool-esp32.ino#LL124C9-L124C9)

i get no logs, nothing, but if i get a multimeter i can see that MISO and MOSI are pulled high, 3.3v and SCK is pulled low, 0v

this might be something wrong with my cc1101, so i will order another one

woahotter, 2023-06-19:
okay, i've debugged some things here and there and found out it was an issue with my wiring - [putting pin 9 on the esp32c3 makes it go into boot mode](https://wiki.seeedstudio.com/XIAO_ESP32C3_Pin_Multiplexing/#d9), and moving those pins to a different area allowed it to boot successfully

also, you were kind of right with the usb to ttl converter, as of currently i am using a usb to ttl coverter to speak with the esp32

[image]

(red bracket are added print statements for debugging)

now to figure out how to speak to the esp32 via usb

mcore1976, 2023-06-21:
_" every time i attempt to transmit tx 010203040506070809aabbccddeeff it stays on Transmitting RF packets.
i think i killed my cc1101 in some other way"_

This looks like CC1101 is not reporting readiness through GDO0 pin state. It may be ether broken or the pin assignment is wrong on this board. Other thing is TTL level 3.3V compatibility. If 5V compatible ESP board then you may have this issues

woahotter, 2023-06-21:
my pin assignment seems to be fine, and the [esp32c3's logic is 3.3v](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/#power-pins)

gdo0 is pulled high 2.9v instead of 3.3v, which doesnt seem like a problem but it might be
ordered another cc1101 anyways

woahotter, 2023-06-21:
okay, fortunately my wiring was correct, but my pin assignments in the code were incorrect

THIS is the only correct pinout diagram, which is meant to be used with the xiao esp32c3
pins 21 and etc. are not in the official pinout, which is weird, only D0, D1 which are not supported by the script

[image]

after editing it in my code, everything worked perfectly

i also edited the setup() function to include `ELECHOUSE_cc1101.getCC1101()` and some other debug stuff

```
void setup() {

  // initialize USB Serial Port CDC
  Serial.begin(115200);
  Serial.println(F("CC1101 terminal tool connected, use 'help' for list of commands...\n\r"));
  Serial.println(F("(C) Adam Loboda 2023\n\r  "));

  Serial.println();  // print CRLF

  Serial.println("cc1101 initializing..\n");

  // initialize CC1101 module with preffered parameters
  cc1101initialize();

  if (ELECHOUSE_cc1101.getCC1101()) {  // Check the CC1101 Spi connection.
    Serial.println("Connection OK");
  } else {
    Serial.println("Connection Error");
  }

  Serial.println("cc1101 initialized\n");

  // setup variables
  bigrecordingbufferpos = 0;
}
```

thank you for your help, everything works now and i can now open car doors and etc.

closed 2023-06-21 by woahotter

mcore1976, 2023-06-22:
Please post a portion of source code with your PIN assignments for this ESP8266 board. It would be very useful for others 👍

woahotter, 2023-06-22:
code pinout:
```
// defining PINs set for ESP32 module
byte sck = 8;
byte miso = 4;
byte mosi = 10;
byte ss = 20;
int gdo0 = 21;
int gdo2 = 7;
```

[image]

MOSI and SCK go to their normal spi positions, except MISO, which goes to pin 4
GDO0 goes to pin 7, and GDO2 goes to pin 21
SS/CSn goes to pin 20

also, if you can, will there be any support for dual cc1101s? ex: one for jamming and one for rx'ing? would be pretty cool, there is some examples out there already with LSatan's library too

mcore1976, 2023-06-23:
Yes, support for dual CC1101 is in my plans. However this is a big change because it requires to add additional parameter for every command. It is not going to be soon but I am thinking of adding such option