mcore1976/cc1101-tool #4: [FEATURE] take advantage of esp8266 webserver

flakka2022, 2023-07-03:
it would be great if be possible to take advantage of esp8266 webserver and have an control panel or web serial comminucate .

mcore1976, 2023-07-05:
I am working on it. Currently waiting for ESP8266 boards delivery.

Egho9, 2023-07-05:
very good project, maybe you can check this ready devices https://item.taobao.com/item.htm?spm=a1z09.2.0.0.44982e8dRdP1Cw&id=645748052953&_u=621lu4d7a43  and  this links https://github.com/joelsernamoreno/EvilCrowRF-V2?spm=a2g0o.detail.1000023.1.32dd3c54j7UiQv to write whole code

mcore1976, 2023-07-06:
I am not using telegram. I am not interested in preparing the code for EvilCrow - it is a different project and has its own dev team. For the ESP8266 my code currently does not handle Watchdog correctly and restarts itself. I will correct it once I get my hands on ESP8266 WEMOS D1 ( it is in delivery now)

flakka2022, 2023-07-06:
> I am not using telegram. I am not interested in preparing the code for EvilCrow - it is a different project and has its own dev team. For the ESP8266 my code currently does not handle Watchdog [...]

> I am not using telegram. I am not interested in preparing the code for EvilCrow - it is a different project and has its own dev team. For the ESP8266 my code currently does not handle Watchdog [...]

good choice to not using telegram . i was about to make another issue to mention restart problem on some fake esp8266 boards and lack of watchdog handling .

mcore1976, 2023-07-08:
I have just received WEMOS D1 mini module and corrected WDT restarts. New source code has been uploaded for ESP8266. It works fine when connected to USB OTG Android phone with Serial Terminal application and Arduino Serial Monitor, however I have some issues when using Putty. Probably some proper Flow Control has to be selected for Serial Port interconnection with ESP8266 boards

flakka2022, 2023-07-08:
> I have just received WEMOS D1 mini module and corrected WDT restarts. New source code has been uploaded for ESP8266. It works fine when connected to USB OTG Android phone with Serial Terminal application and Arduino Serial Monitor, however in Putty - you have to ensure that you are using NONE flowcontrol not the default XON/XOFF flowcontrol in Serial Port parameters of the Putty session. The good is that ESP8266 has plenty of space in FLASH simulated EEPROM - 4096 bytes which gives plenty of possibilities to record longer RF sewuences. For ESP32 it is only 512 bytes just to remind... I will start developing this WIFI serial connectivity stuff - there is good tutorial over here https://www.hackster.io/chuygen/wireless-serial-bridge-d149ef

thanks for being much dedicated to project .
in jamming mode i still have same restart problem . serial log is :
15:51:01.563 -> 
15:51:01.563 -> cc1101 initialized. Connection OK
15:51:01.563 -> 
15:51:06.402 -> jam
15:51:06.402 -> 
15:51:06.402 -> Jamming changed to Enabled
15:51:09.571 -> 
15:51:09.571 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:51:09.571 -> 
15:51:09.571 -> Soft WDT reset
15:51:09.571 -> 
15:51:09.571 -> Exception (4):
15:51:09.571 -> epc1=0x4020454d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
15:51:09.571 -> 
15:51:09.571 -> >>>stack>>>
15:51:09.571 -> 
15:51:09.571 -> ctx: cont
15:51:09.571 -> sp: 3ffffe00 end: 3fffffd0 offset: 0160
15:51:09.571 -> 3fffff60:  0000003c 3ffef6f4 3ffef65c 40202b94  
15:51:09.620 -> 3fffff70:  00000000 fffffffc 00000080 40202f5b  
15:51:09.620 -> 3fffff80:  00000000 3ffe882a 3ffef758 3ffef8c0  
15:51:09.620 -> 3fffff90:  3fffdad0 00000023 3ffef758 3ffef8c0  
15:51:09.620 -> 3fffffa0:  3fffdad0 00000000 3ffef894 3ffef8c0  
15:51:09.620 -> 3fffffb0:  3fffdad0 00000000 3ffef894 402059e8  
15:51:09.620 -> 3fffffc0:  feefeffe feefeffe 3fffdab0 40100e75  
15:51:09.620 -> <<<stack<<<
15:51:09.620 -> 
15:51:09.620 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:51:09.635 -> 
15:51:09.635 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
15:51:09.635 -> 
15:51:09.667 -> load 0x4010f000, len 3424, room 16 
15:51:09.667 -> tail 0
15:51:09.667 -> chksum 0x2e
15:51:09.667 -> load 0x3fff20b8, len 40, room 8 
15:51:09.667 -> tail 0
15:51:09.667 -> chksum 0x2b
15:51:09.667 -> csum 0x2b
15:51:09.667 -> v00048890
15:51:09.667 -> ~ld
15:51:09.732 ->     o s  g| l d`c  |{ d n  g l`  { d d  CC1101 terminal tool connected, use 'help' for list of commands...
15:51:09.732 -> 
15:51:09.732 -> (C) Adam Loboda 2023
15:51:09.732 -> 
  
15:51:09.732 -> 
15:51:09.733 -> cc1101 initialized. Connection OK

mcore1976, 2023-07-08:
try to upload most recent code, the last one works just fine

flakka2022, 2023-07-08:
> try to upload most recent code, the last one works just fine. The reason was too short watchdog time

Unfortunately, the same error persists. To make sure the new code is uploaded, I uploaded the default arduino blank code, and then I uploaded your new recent code from about 20 minutes ago, but the result is as follows.
17:00:46.000 -> jam
17:00:46.000 -> 
17:00:46.000 -> Jamming changed to Enabled
17:00:49.051 -> 
17:00:49.051 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:00:49.084 -> 
17:00:49.084 -> Soft WDT reset
17:00:49.084 -> 
17:00:49.084 -> Exception (4):
17:00:49.084 -> epc1=0x40204571 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
17:00:49.084 -> 
17:00:49.084 -> >>>stack>>>
17:00:49.084 -> 
17:00:49.084 -> ctx: cont
17:00:49.084 -> sp: 3ffffe00 end: 3fffffd0 offset: 0160
17:00:49.084 -> 3fffff60:  0000003c 3ffef6f4 3ffef65c 40202bb3  
17:00:49.084 -> 3fffff70:  00000000 fffffffc 00000080 40202f7f  
17:00:49.084 -> 3fffff80:  00000000 3ffe882a 3ffef758 3ffef8c0  
17:00:49.084 -> 3fffff90:  3fffdad0 00000023 3ffef758 3ffef8c0  
17:00:49.117 -> 3fffffa0:  3fffdad0 00000000 3ffef894 3ffef8c0  
17:00:49.117 -> 3fffffb0:  3fffdad0 00000000 3ffef894 40205a18  
17:00:49.117 -> 3fffffc0:  feefeffe feefeffe 3fffdab0 40100e75  
17:00:49.117 -> <<<stack<<<
17:00:49.117 -> 
17:00:49.117 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
17:00:49.148 -> 
17:00:49.148 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
17:00:49.148 -> 
17:00:49.148 -> load 0x4010f000, len 3424, room 16 
17:00:49.181 -> tail 0
17:00:49.181 -> chksum 0x2e
17:00:49.181 -> load 0x3fff20b8, len 40, room 8 
17:00:49.181 -> tail 0
17:00:49.181 -> chksum 0x2b
17:00:49.181 -> csum 0x2b
17:00:49.181 -> v000488c0
17:00:49.181 -> ~ld
17:00:49.245 -> ����n�s��g|�l�d`c��|{�d�g��g�l`��s�l�l��CC1101 terminal tool connected, use 'help' for list of commands...
17:00:49.245 -> 

17:00:49.245 -> (C) Adam Loboda 2023
17:00:49.245 -> 
  
17:00:49.245 -> 
17:00:49.245 -> cc1101 initialized. Connection OK

mcore1976, 2023-07-08:
hmm it is strange, because I am not getting this on my WEMOS D1 MINI. It firest WDT exactly in 3 seconds...  Looks like on your board the command : "ESP.wdtFeed();" is not working at all ( https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/ ) 

You may try to play with these values :

in setup() part 
     // Enable software watchdog in ESP8266 chip with 3 second fuse in case something goes wrong...
     ESP.wdtEnable(3000); 

in JAM processing part 

      // if jamming mode activate continously send something over RF...
      if ( jammingmode == 1)
      { 
        // populate cc1101 sending buffer with random values
        randomSeed(analogRead(0));
        for (i = 0; i<60; i++)
           { ccsendingbuffer[i] = (byte)random(255);  };        
        // send these data to radio over CC1101
        ELECHOUSE_cc1101.SendData(ccsendingbuffer,60);
        // feed the watchdog
       // you may change following to ESP.wdtEnable(3000)
        ESP.wdtFeed();
      };

flakka2022, 2023-07-08:
a vesrion before of that library was ok for me . should create an issue for SmartRC ?

mcore1976, 2023-07-08:
try to setup 160 MHz speed for ESP8266 then  it works a bit more stable.
Arduino IDE -> Tools /CPU  Frequency / change from 80 MHz to 160 MHz

flakka2022, 2023-07-09:
i think it is possible to take out internal wifi ability and free cpu . i will test and share result here .

flakka2022, 2023-07-09:
No matter what I did, the crash problem was not resolved and it persists (it crashes immediately after entering the jam command). Should I use another board? There is no way apparently.

mcore1976, 2023-07-09:
yes, please try with another board. In my case it crashes from time to time but not frequentlyWysłano z mojego urządzenia Galaxy
-------- Oryginalna wiadomość --------Od: flakka2022 ***@***.***> Data: 9.07.2023  16:52  (GMT+01:00) Do: mcore1976/cc1101-tool ***@***.***> DW: Adam Loboda ***@***.***>, Comment ***@***.***> Temat: Re: [mcore1976/cc1101-tool] [FEATURE] take advantage of esp8266 webserver (Issue #4) 
No matter what I did, the crash problem was not resolved and it persists (it crashes immediately after entering the jam command). Should I use another board? There is no way apparently.

flakka2022, 2023-07-14:
problem solved .

closed 2023-07-14 by flakka2022