mcore1976/cc1101-tool #6: problems of rx , playraw , show and etc in esp8266 boards

flakka2022, 2023-07-14:
facing with following error in commands like rx , playraw and etc .
```
19:30:44.469 -> rx
19:30:44.469 -> 
19:30:44.469 -> Receiving and printing RF packet changed to Enabled
19:30:46.361 -> 
19:30:46.361 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
19:30:46.361 -> 
19:30:46.361 -> Exception (9):
19:30:46.361 -> epc1=0x40208624 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00bcbf65 depc=0x00000000
19:30:46.361 -> 
19:30:46.361 -> >>>stack>>>
19:30:46.361 -> 
19:30:46.361 -> ctx: cont
19:30:46.361 -> sp: 3ffffe00 end: 3fffffd0 offset: 0150
19:30:46.361 -> 3fffff50:  000000d8 00000000 3ffef758 40204e1c  
19:30:46.361 -> 3fffff60:  000000d8 3ffef6f4 3ffef758 4020165c  
19:30:46.393 -> 3fffff70:  00000000 fffffffc 00000080 40202fd3  
19:30:46.393 -> 3fffff80:  00000000 3ffe882a 3ffef758 3ffef8c0  
19:30:46.393 -> 3fffff90:  3fffdad0 00000023 3ffef758 3ffef8c0  
19:30:46.393 -> 3fffffa0:  3fffdad0 00000000 3ffef894 3ffef8c0  
19:30:46.393 -> 3fffffb0:  3fffdad0 00000000 3ffef894 40205a6c  
19:30:46.393 -> 3fffffc0:  feefeffe feefeffe 3fffdab0 40100e75  
19:30:46.393 -> <<<stack<<<
19:30:46.393 -> 
19:30:46.393 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
19:30:46.425 -> 
19:30:46.425 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
19:30:46.425 -> 
19:30:46.457 -> load 0x4010f000, len 3424, room 16 
19:30:46.457 -> tail 0
19:30:46.457 -> chksum 0x2e
19:30:46.457 -> load 0x3fff20b8, len 40, room 8 
19:30:46.457 -> tail 0
19:30:46.457 -> chksum 0x2b
19:30:46.457 -> csum 0x2b
19:30:46.457 -> v00048910
19:30:46.457 -> ~ld
19:30:46.521 -> ����n�s��g|�d�d`c��|{�l�g��o�d`��s�l�l��CC1101 terminal tool connected, use 'help' for list of commands...
19:30:46.521 -> 

19:30:46.521 -> (C) Adam Loboda 2023
19:30:46.521 -> 
  
19:30:46.521 -> 
19:30:46.521 -> cc1101 initialized. Connection OK
19:30:46.521 -> 
```
also 
```
19:33:28.345 -> show
19:33:28.345 -> Wrong parameters.
```

mcore1976, 2023-07-14:
If using commands with RAW like : "rxraw XXX", "recraw XXX", "playraw XXX", "showraw"  when switching to packet mode before using  commands "rx", "tx xxxxxxxxx", "jam" and similar you have to use command "init" first to restart CC1101 board. Otherwise issues like the one above may appear. 

Command "show" will display something only if there is something in the receiving buffer.  In other cases you will get "wrong command" output

BTW. I am not getting this error on my board

flakka2022, 2023-07-14:
thank you for responding . issue solved .
anyway , can you please check [this one ](https://github.com/mcore1976/jammer/issues/3) too ? if you don't mind .

closed 2023-07-14 by flakka2022