mcore1976/cc1101-tool #9: add display

avrilboy, 2023-08-05:
can you add the display and menu like ttgo t display and replay signal easyley or design like fliper zero by display its very dificult to work with this device 

soryy about my request ..ithink its beter to use for users thanks alot

mcore1976, 2023-08-05:
I will think about it, but the point is that not every module, especially ESP32 based like XIAO ESP32C3 has enough pins to connect both CC1101 and OLED module as well.  

Currently my solution was tested with following boards :
- Arduino Pro Micro 3.3V/8MHz
- ESP8266 WEMOS D1 MINI & clones
- ESP32 WROOM module 
- ESP32C3 XIAO SEEDUINO
- Arduino Pro Mini ( ATMEGA328 based) with TTL converter
- Raspbery Pi Pico  RP2060 module

Some of these boards do not support secondary SPI interface to connect OLED/LCD or do not have enough free pins to connect it. 

My goal was to build simplest and cheapset CC1101 tool and I succeeded. If you use chineese CC1101 board for 3$ and ESP8266 WEMOS D1 Mini clone for 2$ you will fit in 5$. 

About usage of this tool. If you have an Android phone and use Serial Terminal application ( https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=pl&gl=US ) 
 to interact with my tool you have possibility to create macros under M1/M2/M3/M4/M5 buttons. These macros can be for example attached to commands : "showraw" , "playraw XXX", "recraw XXX" etc.  Therefore you can record & replay with single click on the screen..
 You should expore this...

BatrakovSV, 2025-05-05:
> I will think about it, but the point is that not every module, especially ESP32 based like XIAO ESP32C3 has enough pins to connect both CC1101 and OLED module at the same time.
> 
> As for now my solution was tested with following boards :
> 
> * Arduino Pro Micro 3.3V/8MHz
> * ESP8266 WEMOS D1 MINI & clones
> * ESP32 WROOM module
> * ESP32C3 XIAO SEEDUINO
> * Arduino Pro Mini ( ATMEGA328 based) with TTL converter
> * Raspbery Pi Pico  RP2060 module
> 
> Some of these boards do not support secondary SPI interface to connect OLED/LCD or do not have enough free pins to connect it.
> 
> My goal was to build simplest and cheapset CC1101 tool and I succeeded. If you use chineese CC1101 board for 3$ and ESP8266 WEMOS D1 Mini clone for 2$ you will fit in 5$. Of course you need some Micro USB <-> USB C cable to connect to the mobile phone or Micro USB <-> USB to PC. Also consider that some people are using this device connected to the PC as I am showing in my video with Universal Radio Hacker, therefore display is not needed.
> 
> About usage of this tool. If you have an Android phone and use Serial Terminal application ( https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=pl&gl=US ) to interact with my tool you have possibility to create macros under M1/M2/M3/M4/M5 buttons. These macros can be for example attached to commands : "showraw" , "playraw XXX", "recraw XXX" etc. Therefore you can record & replay with single click on the screen.. You should expore this.. Also you can use phone clipboard to copy and paste sequences from some storage. You won't be able to use it with display only.

What about this?

https://aliexpress.ru/item/1005006968230803.html

[image]