mcore1976/cc1101-tool #10: I want to add a Bluetooth serial port module.

BG7UHS, 2023-08-08:
I want to add a Bluetooth serial module, so that I can get rid of the bondage of USB cable.
[image]
[image]

mcore1976, 2023-08-08:
good idea, but you need o connect to TX/RX lines of your Arduino board ( TXO/RXI lines) to your Bluetooth adapter. 

Also  Arduino Pro Micro you would need to use different Serial Port number in setup() function :  Serial1

https://www.arduino.cc/reference/en/language/functions/communication/serial/

you need to change all the commands in the source code

from **Serial.**XXXXX

to 

**Serial1.**XXXX

like this 

void setup(){ 
    Serial1.begin(9600);
}

void loop(){
    Serial1.print("HelloWorld");
}

It can be done with Find&Replace in the Notepad application..

BG7UHS, 2023-08-08:
Thank you very much. I plan to use rp2040 with better performance.I'll consult you then.

mcore1976, 2023-09-04:
Try ESP8266 wifi version, just posted yesterday. It allows wireless remote telnet connection (f.ex. using ConnectBot on Android)  and has exactly the same features as wired one.

closed 2023-09-04 by mcore1976