mcore1976/cc1101-tool #11: The scan command didn't get any feedback.

BG7UHS, 2023-08-10:
The scan command didn't get any feedback.
Tested several development versions, all of which are the same.
[image]

mcore1976, 2023-08-10:
You shoulkd narrow the band for searching like

scan 314 316 

or even 

scan 314 315

scan 315 316

scan function requires RF signal to be constantly available therefore keyfob must be pressed for at least 30 seconds.

mcore1976, 2023-08-10:
explained above. If looking for wider band I would suggest to use RTL SDR USB dongle and SDRSharp application to find out the peak frequency. Scan function is only for finetuning the frequency spot

closed 2023-08-10 by mcore1976