mcore1976/cc1101-tool #14: Would love to have you work with me to move this stuff over to the Clipper1 with TFT.

OzInFl, 2023-09-11:
Already have it built and fucntioning somewhat, but would love to work with you to envelop your stuff into the GUI..

Here is the repo, LMK if youre interested,,   https://github.com/OzInFl/CLIPPER1

mcore1976, 2023-09-17:
My code requires keyboard to be present to enter CLI commands. How do you plan to port this ? Please elaborate..

OzInFl, 2023-09-17:
LVGL Keyboard can be used for that purpose,  I currently use one for the
frequency input, another can be added that is alphanumeric that can be used
for the CLI    also can use dropdowns to capture pre-canned commands as
well as last x commands.. etc..   I can create the input textbox and output
textbox, etc that can interface with the CLI so instead of Serial.println
it would be like this:
 "lv_textarea_set_text(ui_textarea, variables);"
can basically convert any IO that was done via serial to the LVGL UI
interface.. easy peasy...

What do you think??

mcore1976, 2023-09-17:
Looks reasonable. I am glad to help but I do not have such board so I can only run "dry debugs".

OzInFl, 2023-09-17:
check out the discord and DM me and we can chat more about it. Here is the server link: https://discord.gg/Dmpme8ph88