mcore1976/cc1101-tool #15: I have cc1101 red module name HW863

gg7200, 2023-10-28:
I have this red module hw863 and it's not working with board I tried esp32 nodemcu arduino pro micro, nano but not working I also change cc1101 module but again it's not working please help if u know why it is not woking

mcore1976, 2023-10-28:
You have full description of this module in this document : https://github.com/mcore1976/cc1101-tool/blob/main/AS07-M1101S.pdf  
the only difference is the naming of MISO pin which is GDO1 here.  Others already confirmed it is working so you must have messed your cabling.
It may happen that your HW863 board is broken as well.

closed 2023-10-28 by mcore1976

AlistairXX, 2024-02-10:
How did you connect its headers. Because they are very tiny. Only possible solution I see is unsoldering them and attaching wires.