mcore1976/cc1101-tool #16: NOTReceiving and printing RF packets

gc9n, 2023-11-03:
Hi there , 
I use a cc1101 and a ESP32 WROOM , if set it to Scanner mode , i get the below messages
Signal found at  Freq: 433.94 Rssi: -34
Signal found at  Freq: 433.92 Rssi: -35
Signal found at  Freq: 433.91 Rssi: -37
Signal found at  Freq: 433.93 Rssi: -36
The problem is that i cannot get the actual packet and see it , save it etc. How can i do that , 
i type 

SETMHZ 433.93
RX

but nothing 

Also theres somehting to mention , i used the cc1101 with a Arduino nano before ESP32  , and i didnt use logic converter
did it Fry my cc1101?   ,  the reason i am asking is beccause i get the results on scanner

mcore1976, 2023-11-03:
Command RX is used for packet mode. In this mode you need to know what preamble shall be used, what address etc..

try command  
setmhz 433.93
rxraw  XXX

where XXX is sampling interval in microseconds.  

By default rxraw uses modulation type ASK.  You may change it with setmodulation X   where X : 0 - 4

gc9n, 2023-11-03:
wow it worked..!

the printing is kinda wierd though is printing even the 000000 the empty stuff.  
and when i got actual data is still printing them into the same raw.   i can fix that

But hey , i didnt fry it , it works! thanks for your quick response

gc9n, 2023-11-03:
REC
PLAY 

sends it to Panic mode restart

gc9n, 2023-11-03:
i see thanks again , very useful info.
i dont understan the RX command ,  cause i dont get any feed back ,  only on the rxraw i get data.

mcore1976, 2023-11-07:
In packet mode (not in RAW mode) the CC1101 chip uses packet adressing and channel numbers. Even though many CC1101 devices are operating on the same radio channel they can have different addresses (like IP address in the internet) and receive only packets that are sent to its own assigned address.  This is to support CSMA/CD type of transmissions over the same radio frequency : https://en.wikipedia.org/wiki/Carrier-sense_multiple_access_with_collision_detection

closed 2023-11-07 by mcore1976