mcore1976/cc1101-tool #18: asciitohex adjust to end string correctly

jps1x2, 2023-11-22:
HI mcore1976!

Function asciitohex is finishing string 2 chars after last written hex value. I am playing with your code and making some adjustements.

File: cc1101-tool-esp32-wroom.ino
Line: 103 hex_ptr[(2*i)+2] = '\0' ;
Correction: hex_ptr[(2*i)] = '\0' ;

Reason: variable i exits incremented from previous for bucle.
When code reaches line 103, i variable has the needed value to finish string fine adding \0, it is not needed to add 2 more char positions to it.

Regards!

Jps1x2

mcore1976, 2024-02-04:
I did not test it but I used this line 103 to put NULL into the string (array of chars) just in case

naldoxable, 2024-02-04:
I can confirm that the correction is **safe**.

A similar situation arises in the last line of **hextoascii**:
`ascii_ptr[i++] = '\0';`
can be modified to:
`ascii_ptr[i] = '\0';`

The case I mentioned is more of a formality, while the solution for the **asciitohex** function is an improvement as it prevents the addition of a blank space.