mcore1976/cc1101-tool #27: Guru Meditation Error: Core  1 panic'ed

MindaugasRam, 2024-02-28:
Hi All,

need some help here

while issuing scan esp32 gets rebooted with panic. 

09:36:49.597 -> scan
09:36:49.597 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
09:36:49.597 -> Core 1 register dump:
09:36:49.597 -> PC      : 0x400d7a5a  PS      : 0x00060530  A0      : 0x800d86e3  A1      : 0x3ffb1e70  
09:36:49.597 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x0000000d  
09:36:49.597 -> A6      : 0x0000002b  A7      : 0x3ffb80c0  A8      : 0x0000002d  A9      : 0x00000009  
09:36:49.635 -> A10     : 0x3ffb8058  A11     : 0x00000000  A12     : 0x3ffb8218  A13     : 0x00000000  
09:36:49.635 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
09:36:49.635 -> EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffa  
09:36:49.635 -> 
09:36:49.635 -> ELF file SHA256: 0000000000000000
09:36:49.635 -> 
09:36:49.635 -> Backtrace: 0x400d7a5a:0x3ffb1e70 0x400d86e0:0x3ffb1f20 0x400d79a9:0x3ffb1f40 0x400d1d06:0x3ffb1f60 0x400d5771:0x3ffb1fb0 0x400861bd:0x3ffb1fd0
09:36:49.673 -> 
09:36:49.673 -> Rebooting...
09:36:49.673 -> ets Jun  8 2016 00:22:57

mcore1976, 2024-02-28:
scan needs  parameterstry scan 433.5 434and read caredully what help commands shows before posting an issueWysłano z mojego urządzenia Galaxy
-------- Oryginalna wiadomość --------Od: MindaugasRam ***@***.***> Data: 28.02.2024  10:39  (GMT+01:00) Do: mcore1976/cc1101-tool ***@***.***> DW: Subscribed ***@***.***> Temat: [mcore1976/cc1101-tool] Guru Meditation Error: Core  1 panic'ed (Issue #27) 
Hi All,
need some help here
while issuing scan esp32 gets rebooted with panic.
09:36:49.597 -> scan
09:36:49.597 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
09:36:49.597 -> Core 1 register dump:
09:36:49.597 -> PC      : 0x400d7a5a  PS      : 0x00060530  A0      : 0x800d86e3  A1      : 0x3ffb1e70
09:36:49.597 -> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x0000000d
09:36:49.597 -> A6      : 0x0000002b  A7      : 0x3ffb80c0  A8      : 0x0000002d  A9      : 0x00000009
09:36:49.635 -> A10     : 0x3ffb8058  A11     : 0x00000000  A12     : 0x3ffb8218  A13     : 0x00000000
09:36:49.635 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000002  EXCCAUSE: 0x0000001c
09:36:49.635 -> EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffa
09:36:49.635 ->
09:36:49.635 -> ELF file SHA256: 0000000000000000
09:36:49.635 ->
09:36:49.635 -> Backtrace: 0x400d7a5a:0x3ffb1e70 0x400d86e0:0x3ffb1f20 0x400d79a9:0x3ffb1f40 0x400d1d06:0x3ffb1f60 0x400d5771:0x3ffb1fb0 0x400861bd:0x3ffb1fd0
09:36:49.673 ->
09:36:49.673 -> Rebooting...
09:36:49.673 -> ets Jun  8 2016 00:22:57

MindaugasRam, 2024-02-28:
Thanks it worked out. Thought if setmhz is defined, should work. Anyway thanks again

mcore1976, 2024-02-29:
working as designed

closed 2024-02-29 by mcore1976