mcore1976/cc1101-tool #32: .

WR117H, 2024-05-10:
**Hi i have a question. This device can record the raw RF data and play it. When I use it with my garage door, it records and plays the raw RF data properly, allowing me to open the garage door. However, when I try to do the same thing with a car, it doesn't work. I mean, it records and plays the raw data properly, but it won't open the door. What could be the problem?**

mcore1976, 2024-05-11:
Yes, because the car uses one time generated codes also called rolling codes. 
It is widely described over internet, you may want to see this page : https://github.com/CR11CS/RollJam-315MHz-433MHz
Create rolljam attack is easy with my cc1101 tool but you need 2 units. One CC1101 needs to  be tuned sligtly around center frequency of the keyfob and jammer activated, while other CC1101 on other board needs to record the keyfob transmission (you need to set the RX bandwidth to lower values first not to catch jamming signal) 
Then you simply replay original signal after disabling jamming.

WR117H, 2024-05-12:
Thank you brother.

closed 2024-05-21 by WR117H