mcore1976/cc1101-tool #40: Cant build the cc1101-tool-rp2040.ino

cruvs7, 2024-11-04:
I am cannot build the [cc1101-tool-rp2040.ino](https://github.com/mcore1976/cc1101-tool/blob/main/cc1101-tool-rp2040.ino) because i get this error
```
 \AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.2.0\libraries\SPI\src/SPI.h:113:23: note: in expansion of macro 'SPI1'
  113 | extern SPIClassRP2040 SPI1;
      |                       ^~~~

exit status 1

Compilation error: exit status 1
```
i tried changing a few things, but couldnt make it work, chatgpt say that i need to change every SPI1 call on the ELECHOUSE_CC1101_SRC_DRV lib to Spi , but i dont think its the rigth answer

crrodger, 2025-05-18:
Having exactly the same challenge. Cannot provide any more info than that provided above. As an aside I think the pin assignments in the readme on the Github landing page might be wrong. For example the assignment

```
byte mosi = 3;
```
I am sure is connecting to the GND pin on a Pico

mcore1976, 2025-05-19:
I am not an author of RPI 2060 pico version but indeed I can see wrong pin assignmentYou should be using the following PINs  for SPI connectivity :pin 21 as MISOpin 22 as CS / SS pinpin 25 as MOSIpin 24 as SCKWysłano z mojego urządzenia Galaxy
-------- Oryginalna wiadomość --------Od: featurealchemist ***@***.***> Data: 19.05.2025  01:04  (GMT+01:00) Do: mcore1976/cc1101-tool ***@***.***> DW: Subscribed ***@***.***> Temat: Re: [mcore1976/cc1101-tool] Cant build the cc1101-tool-rp2040.ino (Issue #40) 
crrodger left a comment (mcore1976/cc1101-tool#40)
Having exactly the same challenge. Cannot provide any more info than that provided above. As an aside I think the pin assignments in the readme on the Github landing page might be wrong. For example the assignment
byte mosi = 3;

I am sure is connecting to the GND pin on a Pico

mcore1976, 2025-07-20:
I really tried to make RP2040-one / RP2040-zero board working with SmartRC library, however when using Arduino with Earle Philhower core ( https://github.com/earlephilhower/arduino-pico  ,  https://arduino-pico.readthedocs.io/en/latest/  )
 it does not work at all...

I tried a lot of things, even editing the code in the SmartRC library... 

However, there is a person who managed to make it work 👍  I suppose he was using different Arduino Core for RP2040 :

https://github.com/MajedAbouhatab/Replacing-Raspberry-Pi-Zero-W-Gate-Opener-With-XIAO-RP2040

https://www.hackster.io/abouhatab/replacing-raspberry-pi-zero-w-gate-opener-with-xiao-rp2040-9ecd4a

mcore1976, 2025-11-02:
Your ARDUINO tells you that you haven't attached the SmartRC library 
"Compilation error: ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory"

https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries/

It is clearly written in README file to the project, quote 

https://github.com/mcore1976/cc1101-tool

"To use it please download following ZIP library from following github link https://github.com/LSatan/SmartRC-CC1101-Driver-Lib and attach it to the script in Arduino IDE."