mcore1976/cc1101-tool #41: AP mode cannot communicate

F-Mesmer, 2025-01-04:
Hello, I have managed to create an AP network (with password) and it appears to work (ping is good), but cannot communicate with serial apps such as Terminus. Also the Serial Monitor in Arduino IDE is blank. No input works. This should not be a hardware issue as the regular cc1101 tool (not wifi) works fine. What could be the issue? Only Terminus responds to help command but crashes soon after.  I've tried everything and am stuck now. Any help would be appreciated.

BatrakovSV, 2025-06-09:
[image]

mcore1976, 2025-07-20:
Hi,  They changed something in the "WIFI" part of ESP32 / ESP8266 SDK for Arduine therefore I am also experiencing this issue. For the moment I do not know how to fix it...