mcore1976/cc1101-tool #44: ESP32 DevKit V1 and NodeMCU ESP-12E Reboot When Running brute Command in cc1101-tool

BatrakovSV, 2025-06-09:
Hi,
I'm facing an issue with the brute command in cc1101-tool on both an ESP32 DevKit V1 and a NodeMCU ESP-12E. When I run the brute command, the boards immediately reboot instead of completing the operation.
Here’s the output I get when connecting to the device via nc (telnet) at 192.168.1.100:23:

init
CC1101 initialized
echo 0
brute 200 4

Starting Brute Forcing press any key to stop...

Right after this message, the board reboots, and the process stops. Occasionally, I see Brute forcing complete after a delay, but the reboot happens consistently.
Setup details:
Hardware: ESP32 DevKit V1 and NodeMCU ESP-12E with CC1101 module

Connection: SPI pins configured correctly (MOSI, MISO, SCK, CS)

Power: Powered via USB with high-quality cables and stable 3.3V supply

Command tested: brute 200 4 (also tried other small values like brute 500 6)

Environment: Arduino IDE with the latest ESP32 and ESP8266 board libraries

Version of cc1101-tool: [specify the version or commit hash if known, e.g., latest from GitHub]

Steps to reproduce:
Connect to the device via nc 192.168.1.100 23

Run init (confirms CC1101 initialized)

Run brute 200 4

The board outputs "Starting Brute Forcing press any key to stop..." and immediately reboots.

Additional info:
Other commands like scan or receive work fine with the CC1101 module.

I’ve verified SPI connections and ensured a stable power supply (3.3V, >500mA).

The reboot occurs even with minimal bit counts (e.g., 4 bits), suggesting it’s not related to high computational load.

I suspect the issue might be related to SPI handling, buffer issues, or interrupts, as mentioned in similar issues.

Could you help identify why the boards reboot when running the brute command? Is this a known issue with SPI or interrupt handling on ESP32/ESP8266? Any suggestions for debugging or workarounds would be greatly appreciated. I can provide additional logs or test specific configurations if needed.
Thanks for your help!

mcore1976, 2025-07-02:
Yes I know about the issue, it looks like bitbanging commands (bitRead)  for C libraries are having some flaws and leading to memory leak. Frankly speaking no one was using this "brute force" cracking before therefore no one was complaining .

        for (brute = 0; brute < poweroftwo ; brute++)  
           { 
           for(int k = 0; k <  5; k++)  // sending 5 times each code
             {
             for(int j = (setting2 - 1); j > -1; j--)  // j bits in a value brute
               {
                 digitalWrite(gdo0, bitRead(brute, j)); // Set GDO0 according to actual brute force value
                 delayMicroseconds(setting);            // delay for selected sampling interval
               }; // end of J loop
               // watchdog
               yield(); 
               // feed the watchdog in ESP8266
               ESP.wdtFeed();                    
             };  // end of K loop
             // checking if key pressed
             if (Serial.available()) break;
           };   

I think I need to convert this "bitRead" into binary operator not to get this memory leaks, something like this

int variable = 10; // Example variable (binary: 00001010)
int bit_position = 1; // Reading the 2nd bit (0-based index)

// Create the bit mask
int mask = 1 << bit_position; // 00000010

// Perform bitwise AND
int masked_value = variable & mask; // 00000010

// Right shift
int bit_value = masked_value >> bit_position; // 00000001

// bit_value now holds the value of the bit at bit_position

BatrakovSV, 2026-10-20:
Hi @mcore1976,

Thanks for the explanation! 

If the issue is indeed with how the `bitRead` macro expands and handles types in this architecture, would replacing it with direct C++ bitwise operations fix the crash?

Specifically, replacing the old line:
```cpp
digitalWrite(gdo0, bitRead(brute, j));
```

With explicit type casting and standard bit shifting:
```cpp
digitalWrite(gdo0, (int)((((unsigned long)brute) >> j) & 0x01));
```

This avoids the Arduino macro entirely, forces proper type sizing for the compiler, and should prevent the memory flaw/reboot on ESP boards. 

What do you think about this fix?